picoruby/picoruby #31: Theoretical minimum

nateberkopec, 2020-10-12:
What is your estimate for the minimum amount of RAM/ROM required to run mmruby and print "Hello world!"?

hasumikin, 2020-10-12:
Take a look at:
https://github.com/hasumikin/mmruby/wiki/Proposal-for-the-Grant-Program-2020-of-Ruby-Association#comparison-between-mrbc-and-mmrbc-regarding-ram-consumption

It uses less than 12KB RAM to compile `puts “Hello World”`, then mruby/c VM can exec it with about 13 KB RAM or so as long as I remember.
So, 13KB is the minimum consumption regarding RAM.
This figure is measured on 64 bit computer. It will be less if you use 32 bit architecture. Guessing 10 KB.

In terms of ROM, sorry but I still don’t have measurement. I guess it will less than 128 KB though, code size is now increasing because implementing Ruby syntax inevitably leads growing binary.
My final goal is less than 256 KB with enough implementation for microcontroller usage.

nateberkopec, 2020-10-13:
Amazing! Thank you so much for the very detailed answer. The figures are extremely impressive.

closed 2020-10-13 by nateberkopec