picoruby/picoruby #130: How to install picoruby globally?

postmodern, 2023-01-24:
I would like to add support for installing picoruby to [ruby-install](https://github.com/postmodern/ruby-install/issues/433). In order to do this, I need to be able to install `picoruby` into either `/usr/local/` (and comply with the [Filesystem Hierarchy Standard](https://refspecs.linuxfoundation.org/FHS_3.0/fhs/index.html)), or `/opt/rubies/picoruby-3.0.2/`, or `~/.rubies/picoruby-3.0.2/`. There doesn't seem to be any documentation on how to install picoruby outside of the repo, but it seems like it could be possible to copy the `bin/pico*` binaries into a `bin/` directory, or copy the `lib/` directory into a `$(PREFIX)/share/picoruby/` directory.

hasumikin, 2023-02-09:
@postmodern 
You can install picoruby through rbenv:

[image]

Does this help you?

postmodern, 2023-02-09:
@hasumikin not exactly. Instead of just wanting to install picoruby, I am the maintainer of [ruby-install]() and am interested in adding support for [installing picoruby to ruby-install](https://github.com/postmodern/ruby-install/issues/433).

hasumikin, 2023-02-13:
@postmodern 
Ah, I see.

>but it seems like it could be possible to copy the bin/pico* binaries into a bin/ directory

I think this works

hasumikin, 2023-02-24:
Will close this issue unless any comment post in a week

closed 2023-02-24 by postmodern