picoruby/picoruby #145: Syntax Error at case / when statement

kirikak2, 2023-05-20:
I am currently investigating and would like to resolve this issue on my own if possible, but will report back once I have done so.

* If there are more than 21 "when" clauses in a "case" statement, a syntax error is raised; if there are less than 20 clauses, no syntax error is raised.
* If there is no "if" statement before the class, the case statement is syntax OK.
* Only case statements defined within a method will result in a Syntax Error. Only case statements defined in a method will result in a Syntax Error; case statements written directly in a class will not result in a Syntax Error.

# Version
PicoRuby 3.1.0dev (debug build)

# Error case

```
if 'a' == 'b'
end

class TestFailcase
  def eval_value(v)
    case v
    when 0
    when 1
    when 2
    when 3
    when 4
    when 5
    when 6
    when 7
    when 8
    when 9
    when 10
    when 11
    when 12
    when 13
    when 14
    when 15
    when 16
    when 17
    when 18
    when 19
    when 20
    when 21
    end
  end
end
```

kirikak2, 2023-05-20:
I have identified the cause of the problem.
The problem is that the parser consumes the stack for each "when" clause in a "case" statement.

If the stack size exceeds the default limit (default 100), a Stack Overflow occurs and the process is terminated, resulting in a Syntax Error.

I think it is not a rare case to write more than 20 whens in a case statement, so I suggest defining a %stack_size directive in parse.y and putting in a value with a margin (200 or 300).

hasumikin, 2023-05-22:
@kirikak2 Thank you for the report.

Increasing `%stack_size` naturally increases a stack size that is allocated statically and consumes more RAM.
In this situation, I feel like I want to have a conditional build that stack_size set to 100 for the microcontroller build and 1000 (for example) for the host computer build.

But, I found one more thing.
If you write `%stack_size 0`, Lemon will automatically allocate heap memory for the parser's stack instead of using a static stack.
It could successfully compile your example Ruby script with a lot of `case`s. 
I guess this is possibly ideal for the PicoRuby compiler (even if the speed of compilation became slower).

Let me think for a while

kirikak2, 2023-05-26:
Thanks for the reply.
I was able to get my code to work without a Syntax Error by setting the stack_size to 0 as well. Thank you very much.

hasumikin, 2024-09-09:
Fixed

closed 2024-09-09 by hasumikin