picoruby/picoruby #171: [UART] TX of `unit: :RP2040_UART1` does not work

hasumikin, 2024-08-07:
- TX and RX of uart0 work
- RX of uart1 works
- Only TX of uart1, the waveform is different every transmit

aikawaYO, 2024-08-31:
日本語にて失礼します
2点考えられる要因があります

## 1.uart_write_blockingは送信完了までブロックしない
おそらくuart_write_blockingは送信完了まで待つことはしません
参考：https://github.com/raspberrypi/pico-sdk/issues/1274#issuecomment-1450182234
`uart_write_blocking`後に`uart_tx_wait_blocking()`を使用してUARTがbusy状態の間待つようにすると改善するかもしれません
[  UART_write_blocking(unit_num, (const uint8_t *)v[1].string->data, len);](https://github.com/picoruby/picoruby/blob/2cd99babb97020d87c26e4f3ca3c91244596c519/mrbgems/picoruby-uart/src/uart.c#L243)

## 2.スリープ時にuart0のみ動作している
スリープ時にUART0のみ動作が許可されています
https://github.com/picoruby/picoruby/blob/2cd99babb97020d87c26e4f3ca3c91244596c519/mrbgems/picoruby-io-console/ports/rp2040/io-console.c#L41-L45

以上、UART1の送信に関しては`uart_write_blocking`が送信完了まで待たないため、スリープ時にデータの欠損が起こっていると思われます

hasumikin, 2024-09-03:
@aikawaYO Thanks!

closed 2024-09-04 by hasumikin