picoruby/picoruby #184: Issue with RbConfig.ruby Implementation when Cross-compiling with ESP-IDF on macOS

yuuu, 2024-12-14:
## Problem Description  

I am currently working on running PicoRuby on ESP32 using ESP-IDF and encountered a build issue related to `picoruby-metaprog`.  

The problem occurs due to the implementation of `RbConfig.ruby`:  
[metaprog.c#L341](https://github.com/picoruby/picoruby/blob/179936109bb46978f449a0fa8841a1992889632a/mrbgems/picoruby-metaprog/src/metaprog.c#L341)  

## Cause of the Issue  

The current implementation switches behavior based on the host OS used for compilation. However, when cross-compiling with ESP-IDF on macOS, `__APPLE__` is defined. This leads to a compilation error due to the undefined function `realpath()`.  

## Proposed Solution  

To resolve this, I propose adding a custom definition `PICORUBY_METAPROG_DISABLE_RBCONFIG_RUBY`. If this macro is defined, the implementation should behave similarly to the bare-metal environment.  

## Request for Feedback  

If there is a better approach to address this issue, please let me know.  

Thank you!

hasumikin, 2024-12-20:
@yuuu 
According to the documentation:, https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/build-system.html
I guess you can use the `ESP_PLATFORM` macro by patching `metaprog.c`.
I'll wait for your PR

yuuu, 2025-02-03:
@hasumikin 
This Issue is closed as it is not a particular problem at this time.

closed 2025-02-03 by yuuu