picoruby/picoruby #206: Want the option to fix timeout on I2C

risgk, 2025-03-29:
With **R2P2 0.3.0**, using Unit 8Angle for M5Stack, I frequently encounter I2C timeout errors.
<https://docs.m5stack.com/en/unit/8Angle>

```
irb> require 'i2c'
=> true
irb> i2c = I2C.new(unit: :RP2040_I2C1, frequency: 100 * 1000, sda_pin: 6, scl_pin: 7)
=> #<I2C:20011b58>
irb> i2c.write(0x43, 0x10)
=> Timeout error in I2C#write (IOError)
irb> i2c.write(0x43, 0x10)
=> 1
irb> i2c.read(0x43, 1).bytes[0]
=> Timeout error in I2C#read (IOError)
```

Using `rescue` and `retry` works as a workaround to some extent.

Also, the current PicoRuby uses `duration_1byte = 1000000 / frequency * 8 * 4;` to determine the timeout time `duration_1byte * len`, so lowering frequency to `25 * 1000` will help avoid errors.

---

However, the response time of I2C devices varies from device to device, and I would like PicoRuby to have the option to fix the timeout on I2C as well. *This will allow more devices to be used successfully, including Units for M5Stack.*

The following API from Arduino is a good reference, but for now, I think adding an argument `timeout` to `I2C#initialize` will cover many use cases.
<https://docs.arduino.cc/language-reference/en/functions/communication/wire/setWireTimeout/>

How about a value of -1 for compatibility operation and a value of 0 for timeout disabled?

The default value should be -1 for compatibility, but a default value of 25000 (25ms), which is common, would make it easier to use in electronic work.

---

I have modified **R2P2 0.2.1** to set the timeout value to `25000` in the `I2C_write_timeout_us()` and `I2C_read_timeout_us()` calls to improve control of Unit 8Angle for M5Stack.

Sorry, but I have not succeeded in building R2P2 0.3.0 or later and have not been able to test it...

yuuu, 2025-04-20:
@risgk 
Thank you for the suggestion. I have discussed this matter with @hasumikin 

As a conclusion, it seems best to allow the I2C#new method to accept a keyword argument timeout, enabling the timeout to be specified in milliseconds.

They mentioned that they will consider implementing it. We might ask for your assistance with testing once the implementation is ready.

risgk, 2025-04-20:
@yuuu @hasumikin 
Thank you for the discussion. Understood. Best regards.

hasumikin, 2025-04-21:
Let me talk to mruby/c core members regarding this. Be patient a little!

hasumikin, 2025-04-28:
After some consideration, I think that introducing `I2C.duration_1byte=` method would solve the problem. This means the `I2C` class exposes the `@duration_1byte` instance variable.
For example, this code doubles the timeout length:
```ruby
i2c = I2C.new(...)
i2c.duration_1byte *= 2
```
This idea doesn't change the `I2C.new` standardized spec, and the existing app code doesn't need fixing.

Regarding disabling the I2C timeout, I disagree because it possibly causes a deadlock.
What do you think? @risgk

risgk, 2025-04-30:
> After some consideration, I think that introducing `I2C.duration_1byte=` method would solve the problem. This means the `I2C` class exposes the `@duration_1byte` instance variable. For example, this code doubles the timeout length:

I think that's a good idea.

Given that Arduino and MicroPython specify timeouts in absolute time (the defaults are 25 and 50 milliseconds, which is long), it seems a bit unnatural and not beginner-friendly, but there should be no practical problem.

> Regarding disabling the I2C timeout, I disagree because it possibly causes a deadlock.

While it is true that there are I2C devices that take a very long time to respond, if the timeout time is long enough, the timeout disabling feature is not essential. @hasumikin

hasumikin, 2025-05-03:
In contrast to my previous statement, I'm introducing a timeout parameter like MicroPython. The default value is 100(ms).

risgk, 2025-05-03:
With **R2P2_PICO-FLASH_MSC-0.4.1-20250503-3c9855a.uf2**, using Unit 8Angle for M5Stack, I2C timeout errors are no longer occur. I guess that almost all I2C devices are fine with 100 ms timeout.

```
irb> require 'i2c'
=> true
irb> i2c = I2C.new(unit: :RP2040_I2C1, frequency: 100 * 1000, sda_pin: 6, scl_pin: 7)
=> #<I2C:2000be20>
irb> i2c.write(0x43, 0x10)
=> 1
irb> i2c.read(0x43, 1).bytes[0]
=> 173
```

I also confirmed that `timeout: 0` does not allow good communication.

```
irb> i2c.write(0x43, 0x10, timeout: 0)
=> Generic error in I2C#write (IOError)
irb> i2c.read(0x43, 1, timeout: 0).bytes[0]
=> Generic error in I2C#read (IOError)
irb> i2c = I2C.new(unit: :RP2040_I2C1, frequency: 100 * 1000, sda_pin: 6, scl_pin: 7, timeout: 0)
=> #<I2C:2001306c>
irb> i2c.write(0x43, 0x10)
=> Generic error in I2C#write (IOError)
irb> i2c.read(0x43, 1).bytes[0]
=> Generic error in I2C#read (IOError)
irb> i2c.write(0x43, 0x10, timeout: 100)
=> 1
irb> i2c.read(0x43, 1, timeout: 100).bytes[0]
=> 174
```

All looks OK to me. Thank you for adding timeout parameters to I2C! @hasumikin @yuuu

closed 2025-05-03 by risgk