picoruby/picoruby #212: How about running RP2040 at 200 MHz clock?

risgk, 2025-04-20:
Raspberry Pi Pico SDK 2.1.1 (2025-02-19) officially allows RP2040 to run at 200MHz clock.

https://github.com/raspberrypi/pico-sdk/releases/tag/2.1.1
> RP2040 has now been certified to run at a system clock of 200Mhz when using a regulator voltage of at least 1.15 volts.

How about running RP2040 with PicoRuby at 200MHz clock?
Might be a good idea to make it optional for compatibility.

I believe RP2040 is currently running at the SDK default of 125MHz.
(If it is running at 133MHz, it is due to my lack of understanding of the code, etc.)
A **1.6x (or 1.5x) improvement** in CPU performance would be nice.

If you have already discussed this issue, I'm sorry, but I missed it.

hasumikin, 2025-04-21:
That deserves to be discussed. At the same time, power consumption is also a concern. Clock-up will need more electric power.
I guess, lower clock should be the default for R2P2. If a hacker who develops a kind of game or something like that, he may want to increase the clock.
So maybe we need to implement a feature that changes the clock at runtime. It may look like `Machine.set_sys_clock`.
But to do this, we will need careful design and tests. What do you think?

risgk, 2025-04-21:
The feature to change the clock at runtime would be realized with `set_sys_clock_khz()` API in Raspberry Pi Pico SDK.
However, this API must basically be called before any initialization process that depend on `clk_sys`. It is not easy to ensure this.

I think `set_sys_clock_khz(200000)` should be called only once at the beginning of `main()` etc.
The following code might be helpful:
https://github.com/earlephilhower/arduino-pico/blob/8d58a9207f0a2665f53c5658250c349ff32939d6/cores/rp2040/main.cpp#L113

As for R2P2, while not elegant, how about offering a 200MHz clocked release binary in addition to the 125MHz clocked one?